Nullkraft/Keypad #1: #DEFINE collisions with other libraries

alastaira, 2019-10-22:
Hi,

key.h specifies a few very generically named defines:
#define OPEN LOW
#define CLOSED HIGH

These definitions of "OPEN" and "CLOSED" can collide when those same common terms are used in other libraries (e.g. in ESP8266WiFi.h, as I spent a few hours this morning discovering!). Are they really necessary - can't LOW and HIGH be used in the code directly? Or could they be renamed to KEYPAD_OPEN and KEYPAD_CLOSED instead perhaps?

Nullkraft, 2019-10-22:
Hello Alastaira,
I was going to suggest bracketing the #defines with #ifndef's.  So:

#ifndef OPEN
#define OPEN LOW
#endif

#ifndef CLOSED
#define CLOSED HIGH
#endif

But I just looked at the ESP8266WiFi source and it seems that they used const.  So that's off the table.  It looks like your suggestion may the best way to go.  So let's try:

#define KEY_OPEN LOW
#define KEY_CLOSED HIGH

I think it will still be descriptive without getting too carried away.  If you could make the changes I would greatly appreciate it.  I'm working on another project right now.  If you want full access to the repository just let me know and I can add you.

Best regards,
Mark

merwok, 2020-06-11:
Hello, I have opened a PR for this.

Nullkraft, 2020-06-11:
Thank you Éric, I replied to your pull request and pointed out a mispelled word in the message. The word is "constnats". If you could fix that I will approve the pull request. If you don't then I will fix it after I do the pull request. I expect to approve the request tomorrow but I can wait longer if you would like me to.

 

Best regards,

Mark Stanley

 
 

Sent: Wednesday, June 10, 2020 at 8:34 PM

PR #6 (2020-06-14): Rename constants to avoid collisions
Fixes #1
closed 2020-06-14 by Nullkraft