Nullkraft/Keypad #2: Update the platform.io library

dzervas, 2020-04-24:
Hello, can you please send the latest 3.2.0 version to platform.io? I need the stated key listener!

Very good job by the way! It's just what I needed! :)

dzervas, 2020-04-24:
I can write a github action to do that when a new git tag is pushed if you like!

dzervas, 2020-04-24:
BTW for anyone else that has the same problem:
`pio lib install https://github.com/Nullkraft/Keypad.git`

Nullkraft, 2020-04-24:
Alright, I just went to the platformio.org website and I will sign up and see what I can accomplish.  I can't promise how long it will take so please be patient with me.

Nullkraft, 2021-02-16:
dservas, you said <<I can write a github action to do that when a new git tag is pushed if you like!>>  I'm not sure if that was in answer to my previous statement but I'm going to assume it was and just let you know that I finally pushed a new tag for the 3.2.0 version of the library with the Stated Key Listener.  I know it's been awhile but if you get this please let me know if I guessed correctly or not.

softegg, 2021-02-16:
I wrote the stated key listener, but I don't know anything about platformio. I would have liked to add it to the official arduino version, but I was told they weren't taking updates. Somebody else seems to have managed it tho. 
When I asked them about it, they said I should merge my changes into their source, even though this is the baseline.
I don't really understand how this is supposed to happen. 
My changes are here, feel free to use them.

Nullkraft, 2021-02-16:
I'm sorry softegg, I was trying to contact dzervas. He was the one who asked for the platform.io update.

But I'm curious what you mean by "they weren't taking updates."  Is that the Arduino Library Manager you are talking about?

softegg, 2021-02-16:
Yeah. Arduino Playground or whatever.
What good is open-source if it is impossible to contribute to it?

Nullkraft, 2021-02-16:
True. Even though I knew it was because of the company splitting I still couldn't let go of the anger. The only saving grace through this whole ordeal is that I finally have full control over my library again and don't have to rely on their website for distribution. Even if (when?) GitHub loses their minds I will always have full control over my own local repo. I can move to any other provider at a moments notice or even stand up my own server. ;)

per1234, 2021-02-16:
It actually didn't have anything to do with the "Arduino vs Arduino" thing. You can read the justifications for making the Arduino Playground read-only here:
https://forum.arduino.cc/index.php?topic=580991.0