Nullkraft/Keypad #7: Ghosting problem on multiple key pressed

pablo-johnson, 2020-06-26:
Hi, Nice work with the library.
I am trying to do a sum of the values of multiple keys pressed at the same time and when im pressing a combination of keys that repeats a column and a row i start to receive ghost events of pressed keys.
Here is my code

`#include <Keypad.h>

const byte ROWS = 4; //four rows
const byte COLS = 4; //four columns
//define the cymbols on the buttons of the keypads
char hexaKeys[ROWS][COLS] = {
  {'-', '-', '-', '-'},
  {'0', '-', '-', '1'},
  {'2', '-', '-', '3'},
  {'4', '-', '-', '5'}
};
byte rowPins[ROWS] = {6, 7, 8, 9}; //connect to the row pinouts of the keypad
byte colPins[COLS] = {2, 3, 4, 5}; //connect to the column pinouts of the keypad

bool bools[6];

//initialize an instance of class NewKeypad
Keypad customKeypad = Keypad( makeKeymap(hexaKeys), rowPins, colPins, ROWS, COLS);

void setup() {
  Serial.begin(9600);
  customKeypad.addStatedEventListener(keypadEvent);
  customKeypad.setHoldTime(500);
}

void loop() {
  char customKey = customKeypad.getKey();

  if (customKey) {
    //    Serial.println(customKey);
  }
  int seconds = 30;
  while (seconds > 0) {
    delay(1000);
    seconds--;
    customKeypad.getKeys();
    float sum = getSum();
    if(sum>0){
      Serial.println(" Sum = " + String(round(sum)));
    }
  }
}

float getSum() {
  float sum = 0;
  for (int i = 0; i < 6; i++) {
    if (bools[i]) {
      //      Serial.print(String(pow(2, i)));
      Serial.println(String(i));
      sum += pow(2, i);
    }
  }
  //  Serial.println("Sum = " + String(sum));
  return sum;
}

void keypadEvent(KeypadEvent key, KeyState state) {
  switch (state) {
    case PRESSED:
      break;
    case RELEASED:
      bools[int(key) - '0'] = false;
      break;
    case HOLD:
      bools[int(key) - '0'] = true;
      break;
    case IDLE:
      bools[int(key) - '0'] = false;
      break;
  }
}`

in order to reproduce the bug try this combination ('B', 'A', '1').
B and A in the same column and A and 1 in the same row.

pablo-johnson, 2020-06-26:
BTW, im working with this keypad
https://www.parallax.com/sites/default/files/downloads/27899-4x4-Matrix-Membrane-Keypad-v1.2.pdf

softegg, 2020-06-26:
Yeah, that is a thing that happens with keypads that don't have diodes in them. Nothing the library can do about it.
Here is a good article that mentions the problem:
http://blog.komar.be/how-to-make-a-keyboard-the-matrix/

-- Timon --

Nullkraft, 2020-06-26:
Hi Pablo,

Thank you for your kind words. :)

Ghosting is a known problem with all keyboards. Most people never see it but their desktop keyboards have the same problem. Keyboard manufacturers solve the problem by adding diodes across some, or all, of the keys. If you ever look up the specification for a manufacturers keyboard you'll see a term called n-key rollover. That tells you how many keys you can press before you start to see ghosting.

If you can't install the diodes inside your keypad then you can mount them externally across the row and column wires instead. Arranging them properly can be very tedious. If you google it you can find *many* examples of how to add the diodes.

That should fix the ghosting problem. But I see that you have a delay(1000) in the sketch which is going to cause you problems. I can help with that too if you would like. But fix the ghosting first.

Nullkraft, 2020-06-26:
Hi Timon,
Thanks for the link. I thought I was going to have to hunt something down myself. ;)

pablo-johnson, 2020-07-12:
> Hi Pablo,
>
> Thank you for your kind words. :)
>
> Ghosting is a known problem with all keyboards. Most people never see it but their desktop keyboards have the same problem. Keyboard manufacturers solve [...]

Hi Timon, 
thanks for your reply and sorry for my late one.
I am implementing a kind of braille training game and that's basically the reason of the delay (waiting for the user input). Finally i ended up using 6 separate buttons instead of this keypad due to this ghost issue and it worked as expected.
Anyway, thanks for your help.

closed 2020-07-12 by pablo-johnson