Nullkraft/Keypad #9: Output high before input in scanKeys

jamesmyatt, 2020-07-12:
Why is there a pin_write(HIGH) before pin_mode(INPUT) in scanKeys?

https://github.com/Nullkraft/Keypad/blob/e20e86e45c99953892459e3f12a4c8dae2242f7a/src/Keypad.cpp#L97

Nullkraft, 2020-07-12:
Hello jamesmyatt,
  In May of 2012, over on the Arduino Forums, [Nick Gammon noted](https://forum.arduino.cc/index.php?topic=107430.msg807100#msg807100) that there was a bug in the library that was causing physical shorts between the column pins. This would happen anytime more than one key was pressed at a time.

  I asked him about it and learned that you have to set each of the non-selected column pins into a "high impedance input" state to prevent them from shorting to the selected column pin.

  Although it's not very clear the Arduino Reference covers this topic in ['digital pins'](https://www.arduino.cc/en/Tutorial/DigitalPins). If you read the entire page you'll notice that this is intended for older versions of the Arduino IDE. I feel it is necessary to maintain that line of code because the keypad library is used by other platforms like the TI-MSP430, ESPxxx and various others on Platformio.org, etc.

jamesmyatt, 2020-07-15:
It's clear that it's best to set it to INPUT when you're done, but why is it also set to HIGH immediately before that?

Nullkraft, 2020-07-17:
It's not that it's **best**, which of course does matter, but that it is used to turn on the internal pullup resistor.

From the Arduino Reference page on Digital Pins:
**_an output pin that is left in a HIGH state will have the_** [internal]  **_pullup resistors set if switched to an input with pinMode()._** 

So while the pinMode of the column pin is still an output I set it HIGH to meet the requirement for restablishing the internal pullup resistor.

The reason I did this was to prevent internal short circuits when more than one button is pressed _at the same time_. This is critical in an application that reads multiple key presses by default but is also necessary if you are only reading single key presses because you just don't know what the user will do. Have **you** ever pressed as many keys as you possibly could just to see what would happen? ;)

The more technical explanation is that the built in pullup resistors are controlled by the same internal registers that are used by digitalWrite() for setting the output to HIGH or LOW. Because of that the Arduino developers had to offer another way to reset the input pullup resistors after any pinMode() changes by the sketch's author. They chose to do so in a two step process.

I don't know if that has changed in the later Arduino releases but I still wanted the library to work with a range of other uController boards in addtion to older versions of the Arduino IDE. It may be time to re-evaluate but of course I have a hundred other things I've been working on since then. :)

jamesmyatt, 2020-07-18:
Switching to high input impedance should be enough to prevent any short-circuit whether it's done with the mode INPUT or INPUT_PULLUP. I'm worried that the momentary active high output might cause unintended consequences if the pins are re-used, as you say in the comments.

Nullkraft, 2020-07-19:
The problem is not the momentary high, per se. It has more to do with the timing between column selection/deselection. Believe me, I was quite shocked when I realized what I was doing wrong. After connecting an o'scope I finally got the timing such that all the other columns are turned off (set HIGH) before the next column is turned on (reset LOW). No more shorts.

But I have to admit that if someone on the Arduino forums hadn't pointed it out I would have probably fought with the multi-keypress version for a lot longer. I was primarily looking to turn on the internal pullups rather than solve any shorting problems. My only excuse is that I was getting caught up in all the little things along the way.

closed 2021-02-16 by Nullkraft