Nullkraft/Keypad #10: Online Simulation for Nullkraft Arduino Keypad from Wokwi 😃

Puneethkumarsn, 2020-12-04:
hi There

Greetings from Wokwi Arduino Simulator 🙂

Thank you for the beautiful libraries and examples. 🙏

Here, we have a permanent link for the simulation:
https://wokwi.com/arduino/projects/283991385282970124

It will be our wish to provide more such simulations to all the examples.

Kindly let us know if you have any suggestions.

I hope you find this interesting. We would like to create a PR to update the links in the ReadMe file so that, the users of Nullkraft/keypad library can have a look and also play around the simulations before actually building or using them in their own projects 

Best Regards
Puneeth

WN2A, 2020-12-05:
Hello Puneeth,

Wow! This is really cool. I think it would be a great idea to add the links. I tried the link you provided above and it works really well for the simple case of pressing a key. The only problem is that I couldn't press-and-hold the '*' key to make the led blink. It only blinks a single time when I press the key but it is supposed to blink repeatedly when I'm holding the key down. The sketch is looking for mouse-down, mouse-hold and mouse-up.

I hope you will still send the PR. The current state of your simulation tool is more than good enough for sketch authors to begin using it.

Best regards,
Mark

Puneethkumarsn, 2020-12-05:
Dear Mark

We ar glad you liked it.

We will try to see why the simulation is not blinking the LED when the `*` key is held pressed. I will surely send the PR for this example and other examples as well soon 😀

Thank you 
Puneeth

Puneethkumarsn, 2020-12-05:
hi There, 

the bugs have been squashed.
Kindly check now please

Thank you 
Puneeth

Nullkraft, 2020-12-07:
Wokwi Arduino Simulator permanent link for the simulation merged into the main readme.md file.

closed 2020-12-07 by Nullkraft