Nullkraft/Keypad #12: Update Arduino library

Patrick-Thomas, 2021-02-15:
Hello,

can you update the version on the Arduino Library Manager to the latest 3.2.0 version. My code makes use of the state event listener which isn't in prior versions.

Thanks

Patrick-Thomas, 2021-03-15:
Hi,

thanks for pushing through this update. I just read through the above thread, have you heard anything back from Chris--A?

Thanks

Nullkraft, 2021-03-15:
No response at all.  It took quite a bit of digging and reading through some of his source code files to find the additional email address. But it seems that he doesn't really want to be contacted.  He has a domain but no response from him on his website.  I don't really know what to do because if I change the name of the library it's going to make people question which one is the 'real' lib.

Patrick-Thomas, 2021-03-17:
OK I see the predicament. For now I'm happy to direct people here to download the zip, but if you could keep trying to contact him that would be great. If he really has no interest in maintaining the repo then Arduino should just transfer it over to you.

Nullkraft, 2021-03-17:
It's been almost 4 years that I've been trying to contact him.  Hope is running pretty thin right about now.  The point about his level of interest is, of course, obvious to you and I but it is difficult for the Arduino crew to make that assumption without doing a bit (lot?) of research.  The alternative for them is to blindly trust my word and should they start doing that it will become chaos on the library manager list.

P.S.  Even though my words indicate understanding of the situation you should know that I choked on each and every one of them as I wrote it. :/

Nullkraft, 2022-03-20:
Hi Patrick-Thomas, bad news. I'm going to have to close this issue as the Arduino folks just got back to me and told me to piss off. Based on their attitudes I will never be able to update the Keypad library ever again.  They allowed Chris--A to take control away from me but they claim they are not the arbitrators of library managers.  I disagree but it's over.  My sincerest apologies but all work on the library will have to be provided distributed from here.

closed 2022-03-20 by Nullkraft