Nullkraft/Keypad #16: Long wire

podaen, 2022-04-19:
I having issues with an ESP32 and long wires (-5m) with any key-library by receiving random numbers. I tried to set a pull up but it diden't work. The wires are svv-F2 12x0.8mm. It is a software problem because there is only 0.1 voltage drop. Could it be there is a timing problem? note: Everything works with short wires. 

```
#include <Keypad.h>
#include <Key.h>

const byte ROWS = 4; //four rows
const byte COLS = 3; //three columns
char keys[ROWS][COLS] = {
    {'1','2','3'},
    {'4','5','6'},
    {'7','8','9'},
    {'*','0','#'}
};

byte rowPins[ROWS] = { 17, 16, 4, 2 };
byte colPins[COLS] = { 14, 12, 13 };

Keypad keypad = Keypad(makeKeymap(keys), rowPins, colPins, ROWS, COLS);

void setup() {
    Serial.begin(115200);// 115200);//57600//921600//2000000
    while (!Serial);

    for (byte i = 0; i < (sizeof(rowPins) / sizeof(rowPins[0])); i++) {
    keypad.pin_mode(rowPins[i], INPUT_PULLUP);
    }
    for (byte i = 0; i < (sizeof(colPins) / sizeof(colPins[0])); i++) {
    keypad.pin_mode(colPins[i], INPUT_PULLUP);
    }

    keypad.addStatedEventListener(keypadEvent); // Add an event listener for this keypad
}
void loop() {
    char key = keypad.getKey();

    if (key) {
        Serial.println(key);
    }
    delay(50);
}

// Taking care of some special events.
void keypadEvent(KeypadEvent key, KeyState kpadState) {
    switch (kpadState) {
    case PRESSED:
        if (key == '#') {

        }
        break;

    case RELEASED:
        if (key == '*') {

        }
        break;

    case HOLD:
        if (key == '*') {

        }
        break;
    }
}
```

podaen, 2022-04-19:
Ps: The previous setup with an arduino did work on 5v.

podaen, 2022-04-19:
When I change the ports to

byte rowPins[ROWS] = { 19, 18, 17, 16 };
byte colPins[COLS] = { 33, 32, 35 };

Numbers pressed from 1,2,3,4,5,6,7,8,9,*,0,#

> Opening port
> Port open
> 1
> 1
> 1
> 4
> 2
> 1
> 8
> 2
> 1
> 0
> 2
> 1

very odd

Nullkraft, 2022-04-19:
Hi Podaen,  do me a favor and take a look at this [issue](https://github.com/Nullkraft/Keypad/pull/15).  It looks like the same problem and solution to what you have.  With 5m long wires you may require an even longer scanTime setting.  No matter the result you will still require pullup resisters, through.

You can get the new scanTime setting/variable by downloading the latest source.

podaen, 2022-04-19:
I was testing now with the I2C line and I have no input. Is there someting I have to change within this library for the I2C with my setup?

Nullkraft, 2022-04-19:
I'm not sure how to answer this.  The keypad library does not directly support I2C.  You would have to use something like this: https://github.com/joeyoung/arduino_keypads which inherits the keypad library and supports I2C.  Is that what you are talking about?

Can you provide a copy of your schematic?

podaen, 2022-04-20:
yes that is what I mean. I have an other I2C library. But yes that is what I mean. First I will test with the keyboard and long wire directly connected. If that works with the adjustments you send me, I will test further with the [I2C library](https://github.com/joeyoung/arduino_keypads/tree/master/Keypad_I2C)

I don't have a shematic... SDA is connected to 21 and SCL is connected to 22 on the PCF8574.

```
/*I2C KEYPAD*/
#include <Wire.h>
#include <Keypad_I2C.h>
#include <Keypad.h>
#include <Key.h>
#define I2CADDR 0x20

const byte ROWS = 4; //four rows
const byte COLS = 3; //three columns
char keys[ROWS][COLS] = {
    {'1','2','3'},
    {'4','5','6'},
    {'7','8','9'},
    {'*','0','#'}
};

// Digitran keypad, bit numbers of PCF8574 i/o port
byte rowPins[ROWS] = { 0, 1, 2, 3 }; //connect to the row pinouts of the keypad
byte colPins[COLS] = { 4, 5, 6 }; //connect to the column pinouts of the keypad

//Keypad_I2C kpd(makeKeymap(keys), rowPins, colPins, ROWS, COLS, I2CADDR, PCF8574);
Keypad_I2C keypad(makeKeymap(keys), rowPins, colPins, ROWS, COLS, I2CADDR);

void setup() {
    Serial.begin(115200);
    while (!Serial);

	//Wire.begin(16, 17);
    Wire.begin();
	//Serial.println("clock");
	//Serial.println(Wire.getClock());

	keypad.begin(makeKeymap(keys));
	//keypad.addEventListener(keypadEvent);                                      // Add an event listener.

    keypad.addStatedEventListener(keypadEvent); // Add an event listener for this keypad

    keypad.setHoldTime(50);  //500                                                 // Default is 1000mS
}
void loop() {
    char key = keypad.getKey();

    if (key) {
        Serial.println(key);
    }
    delay(50);
}

// Taking care of some special events.
void keypadEvent(KeypadEvent key, KeyState kpadState) {
    switch (kpadState) {
    case PRESSED:
        if (key == '#') {

        }
        break;

    case RELEASED:
        if (key == '*') {

        }
        break;

    case HOLD:
        if (key == '*') {

        }
        break;
    }
}
```

podaen, 2022-04-20:
This is the code without I2C

```
#include <Keypad.h>
#include <Key.h>

const byte ROWS = 4; //four rows
const byte COLS = 3; //three columns
char keys[ROWS][COLS] = {
    {'1','2','3'},
    {'4','5','6'},
    {'7','8','9'},
    {'*','0','#'}
};

//byte rowPins[ROWS] = { 17, 16, 4, 2 };
//byte colPins[COLS] = { 14, 12, 13 };

//byte rowPins[ROWS] = { 19, 18, 17, 16 };
//byte colPins[COLS] = { 33, 32, 35 };

//byte rowPins[ROWS] = { 19, 18, 17, 16 };
//byte colPins[COLS] = { 25, 26, 27 };

byte rowPins[ROWS] = { 18, 5, 17, 16 }; // GIOP18, GIOP5, GIOP17, GIOP16 connect to the row pins
byte colPins[COLS] = { 4, 0, 2 };  // GIOP4, GIOP0, GIOP2 connect to the column pins

Keypad keypad = Keypad(makeKeymap(keys), rowPins, colPins, ROWS, COLS);

void setup() {
    Serial.begin(115200);// 115200);//57600//921600//2000000
    while (!Serial);

    for (byte i = 0; i < (sizeof(rowPins) / sizeof(rowPins[0])); i++) {
    keypad.pin_mode(rowPins[i], INPUT_PULLUP);
    }
    for (byte i = 0; i < (sizeof(colPins) / sizeof(colPins[0])); i++) {
    keypad.pin_mode(colPins[i], INPUT_PULLUP);
    }

    keypad.addStatedEventListener(keypadEvent); // Add an event listener for this keypad
}
void loop() {
    char key = keypad.getKey();

    if (key) {
        Serial.println(key);
    }
    delay(50);
}

// Taking care of some special events.
void keypadEvent(KeypadEvent key, KeyState kpadState) {
    switch (kpadState) {
    case PRESSED:
        if (key == '#') {

        }
        break;

    case RELEASED:
        if (key == '*') {

        }
        break;

    case HOLD:
        if (key == '*') {

        }
        break;
    }
}
```

podaen, 2022-04-20:
I am not sure how to write a test procedure to get the scan time. Could you help me please?

podaen, 2022-04-20:
I get an error not found, because the .h is not corresponding with .cpp 

.h
`void addOOPEventListener(void (*listener)(void *ObjectContext, char, KeyState), void * ObjectContext);`

.cpp
`void Keypad::addOOPEventListener(void (*listener)(void *context, char, KeyState), void *ObjectContext)`

I can not make a pull because I downloaded manualy. if you change context to ObjectContext the error dissapears.

podaen, 2022-04-20:
Now it works for the keypad with only this library, I played with the parameter 

keypad.setScanTime(10);

And I changed column pins to output. That was my mistake.

```
#include <Keypad.h>
#include <Key.h>

const byte ROWS = 4; //four rows//input
const byte COLS = 3; //three columns//output
char keys[ROWS][COLS] = {
    {'1','2','3'},
    {'4','5','6'},
    {'7','8','9'},
    {'*','0','#'}
};

byte rowPins[ROWS] = { 18, 5, 17, 16 }; // GIOP18, GIOP5, GIOP17, GIOP16 connect to the row pins
byte colPins[COLS] = { 4, 0, 2 };  // GIOP4, GIOP0, GIOP2 connect to the column pins

Keypad keypad = Keypad(makeKeymap(keys), rowPins, colPins, ROWS, COLS);

void setup() {
    Serial.begin(115200);
    while (!Serial);

   keypad.begin(makeKeymap(keys));
   keypad.setScanTime(10);

    for (byte i = 0; i < (sizeof(rowPins) / sizeof(rowPins[0])); i++) {
    keypad.pin_mode(rowPins[i], INPUT_PULLUP);
    }
    for (byte i = 0; i < (sizeof(colPins) / sizeof(colPins[0])); i++) {
    keypad.pin_mode(colPins[i], OUTPUT);
    }

    keypad.addStatedEventListener(keypadEvent); // Add an event listener for this keypad
}
void loop() {
    char key = keypad.getKey();

    if (key) {
        Serial.println(key);
    }
    delay(10);
}

// Taking care of some special events.
void keypadEvent(KeypadEvent key, KeyState kpadState) {
    switch (kpadState) {
    case PRESSED:
        if (key == '#') {

        }
        break;

    case RELEASED:
        if (key == '*') {

        }
        break;

    case HOLD:
        if (key == '*') {

        }
        break;
    }
}
```

podaen, 2022-04-20:
Now the I2C works too, by added the setscantime().

podaen, 2022-04-24:
After testing a lot, it seems not be stable. Soo I went for looking how it could be solved.

1st without I2C it seems to work by setting the setscantime.
2th I made some tasks for the I2C to handle other intensif tasks that aquire a lot of cpu.

But after some time it been clear that there is still an issue. Mayble floats of the signal because the expander module doesn't have pullup resistors or maybe because the scantime doesn't effect on I2C.

Because I have long wires I went to lookup what is the maximum length of an I2C device. And than I was stumpt to know that for a normal cable it isn't that much... 2.25m!!! that's halve of what I have. So continuing on using IC2 is not an option anymore.
https://electronics.stackexchange.com/questions/106265/maximum-i2c-bus-length

podaen, 2022-04-24:
I changed it back to directly connected. Now I have that same key that is annoying (*). The key get suddenly active after pressing a serie of numbers. Very strange because directly connected instread of I2C...

podaen, 2022-04-24:
Changed to other pins, problem solved now!

closed 2022-04-25 by podaen

podaen, 2022-09-09:
> Because I have long wires I went to lookup what is the maximum length of an I2C device. And than I was stumpt to know that for a normal cable it isn't that much... 2.25m!!! that's halve of what I have. So continuing on using IC2 is not an option anymore.

I could not let go and now I have it working for over a month. So don't let you be scared of the length.

Nullkraft, 2022-09-09:
This is wonderful news. I don't suppose you could let us know which Arduino and which pins you are using?

podaen, 2022-09-09:
No problem. I think I did start with other I2C pins than the default.

> Arduino IDE 1.8.19
> Pins 21,22
> SDK: ESP Arduino 2.0.4
> Board ESP32 DevKit Wrover IE

> ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 3, 4MB external flash
> Chip id: D4BD9710
> Model number: ESP32-D0WDQ5

Additional tip: 

- Don't place the keypad in a separate task. At the end it will give you synchronization trouble with other task. You need to put a lot of afford in and give little advantage, even when you execute them whit a time interval. You can make it work, but you will have always issues when you change something after in you code... At least in my case it was.

- Don't set setScanTime if you are using I2C

```
void IniKeyPad() {
	//Wire.begin(33, 32); //(16, 17);
	Wire.begin(21, 22);//SDA, SCL (standaard)

	keypad.begin(makeKeymap(keys));

	for (byte i = 0; i < (sizeof(rowPins) / sizeof(rowPins[0])); i++) {
		keypad.pin_mode(rowPins[i], INPUT); //INPUT_PULLUP
	}
	for (byte i = 0; i < (sizeof(colPins) / sizeof(colPins[0])); i++) {
		keypad.pin_mode(colPins[i], OUTPUT);
	}
	keypad.addEventListener(keypadEvent); // Add an event listener for this keypad.
	keypad.setHoldTime(500);//time holled keypress
}
```

I hope this helps.